Aus3D/RUMBA32 #8: PWM for Bltouch and LEDs on AUX pins?

wesk-ms, 2019-04-17:
Hello,

I built a Rumba32 board a couple months ago and it's been outstanding.  Very nice design...great work.

I'm trying to add a Bltouch and a string of neopixel LEDs to my corexy printer and not having any success.  I'm not seeing any PWM output on the pins PWM_1 or PWM_2.  I'm building a new build of Marlin every 3-4 days.  Also using the STM32Duino core and tools that I downloaded about a month ago.

The printer uses 6 steppers (tmc5160 stepsticks).  Bltouch device is genuine and v2.2.  It performs the power on self test correctly, but I'm not able to actuate the plunger via the LCD or with Gcode.

I'm really more concerned about getting the Bltouch working.  The pins I've tried for SERVO0_PIN are PD15, PD14. PD13, and PD12.  I've also tried simply specifying 62, 61, 60, and 59 in the pins_RUMBA32.h file, as well.

Something I found interesting...  I'm using VS Code to build Marlin, and Intellisense resolves PD15 to pin 63, and PD14 to 62, etc...  This is 1 pin off of the physical CPU connections.  I'm thinking that this has to be something funky with #includes since everything else on the board works flawlessly as is, despite Intellisense being 1 pin off for all 15 to 20 pins that I bothered to check.

I've looked through the timer definition code for this variant and I'm just not seeing any obvious errors.

--Any ideas what I might be overlooking?
--Have you or anyone else been successful in getting a Bltouch or LEDs to work on these pins?
--Is there a different pin I should be using for the BLtouch?

Thanks!
Wes

WK-MS, 2019-04-17:
@chrissbarr Do you have any thoughts as to why I'm unable to get a Bltouch working on the Rumba32 board on pins PD15, PD14, PD13, or PD12?

Thank you!

wesk-ms, 2019-04-28:
@chrissbarr , bumping this one again.  Would really like to get your insight on wether or not a timer change is needed to enable pwm output on PD15, PD14, PD13, and PD12.  I'll also go ahead and post this same question in the stm32duino repo.

Thanks!

chrissbarr, 2019-04-29:
Hi Wes,

Sorry that I haven't replied sooner. Too many notifications on GitHub - I missed your earlier messages. My apologies.

Very happy to hear that you've put together one of these boards and are enjoying it! That's great news. Glad that it is working well for you. 

At the moment, I don't believe that PWM_1 or PWM_2 are configured to have PWM enabled in the stm32duino core. The relevant place where the PWM is configured is here:

https://github.com/stm32duino/Arduino_Core_STM32/blob/master/variants/RUMBA32_F446VE/PeripheralPins.c

PWM pins are defined from line 125 onwards:

```
const PinMap PinMap_PWM[] = {
  {PA_0,  TIM2,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM2, 1, 0)},  // TIM2_CH1
  //  {PA_0,  TIM5,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM5, 1, 0)},  // TIM5_CH1
  //  {PA_1,  TIM2,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM2, 2, 0)},  // TIM2_CH2
  {PA_1,  TIM5,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM5, 2, 0)},  // TIM5_CH2
  //  {PA_2,  TIM2,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM2, 3, 0)},  // TIM2_CH3 - STLink Tx
  //  {PA_2,  TIM5,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM5, 3, 0)},  // TIM5_CH3 - STLink Tx
  //  {PA_2,  TIM9,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF3_TIM9, 1, 0)},  // TIM9_CH1 - STLink Tx
  //  {PA_3,  TIM2,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM2, 4, 0)},  // TIM2_CH4 - STLink Rx
  //  {PA_3,  TIM5,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM5, 4, 0)},  // TIM5_CH4 - STLink Rx
  //  {PA_3,  TIM9,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF3_TIM9, 2, 0)},  // TIM9_CH2 - STLink Rx
  {PA_5,  TIM2,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM2, 1, 0)},  // TIM2_CH1
  //  {PA_5,  TIM8,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF3_TIM8, 1, 1)},  // TIM8_CH1N
  {PA_6,  TIM13,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF9_TIM13, 1, 0)},  // TIM13_CH1
  //  {PA_6,  TIM3,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM3, 1, 0)},  // TIM3_CH1
  {PA_7,  TIM14,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF9_TIM14, 1, 0)},  // TIM14_CH1
  //  {PA_7,  TIM1,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM1, 1, 1)},  // TIM1_CH1N
  //  {PA_7,  TIM3,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM3, 2, 0)},  // TIM3_CH2
  //  {PA_7,  TIM8,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF3_TIM8, 1, 1)},  // TIM8_CH1N
  {PA_8,  TIM1,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM1, 1, 0)},  // TIM1_CH1
  {PA_9,  TIM1,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM1, 2, 0)},  // TIM1_CH2
  {PA_10, TIM1,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM1, 3, 0)},  // TIM1_CH3
  {PA_11, TIM1,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM1, 4, 0)},  // TIM1_CH4
  {PA_15, TIM2,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM2, 1, 0)},  // TIM2_CH1
  //  {PB_0,  TIM1,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM1, 2, 1)},  // TIM1_CH2N
  //  {PB_0,  TIM3,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM3, 3, 0)},  // TIM3_CH3
  {PB_0,  TIM8,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF3_TIM8, 2, 1)},  // TIM8_CH2N
  //  {PB_1,  TIM1,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM1, 3, 1)},  // TIM1_CH3N
  //  {PB_1,  TIM3,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM3, 4, 0)},  // TIM3_CH4
  {PB_1,  TIM8,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF3_TIM8, 3, 1)},  // TIM8_CH3N
  {PB_2,  TIM2,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM2, 4, 0)},  // TIM2_CH4
  {PB_3,  TIM2,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM2, 2, 0)},  // TIM2_CH2
  {PB_4,  TIM3,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM3, 1, 0)},  // TIM3_CH1
  {PB_5,  TIM3,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM3, 2, 0)},  // TIM3_CH2
  {PB_6,  TIM4,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM4, 1, 0)},  // TIM4_CH1
  {PB_7,  TIM4,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM4, 2, 0)},  // TIM4_CH2
  //  {PB_8,  TIM10,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF3_TIM10, 1, 0)},  // TIM10_CH1
  //  {PB_8,  TIM2,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM2, 1, 0)},  // TIM2_CH1
  {PB_8,  TIM4,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM4, 3, 0)},  // TIM4_CH3
  {PB_9,  TIM11,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF3_TIM11, 1, 0)},  // TIM11_CH1
  //  {PB_9,  TIM2,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM2, 2, 0)},  // TIM2_CH2
  //  {PB_9,  TIM4,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM4, 4, 0)},  // TIM4_CH4
  {PB_10, TIM2,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM2, 3, 0)},  // TIM2_CH3
  {PB_13, TIM1,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM1, 1, 1)},  // TIM1_CH1N
  {PB_14, TIM12,  STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF9_TIM12, 1, 0)},  // TIM12_CH1
  //  {PB_14, TIM1,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF1_TIM1, 2, 1)},  // TIM1_CH2N
  //  {PB_14, TIM8,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF3_TIM8, 2, 1)},  // TIM8_CH2N
[... cut ...] ow dependent they are on the PWM frequency or how the stm32duino PWM frequency compares to the Arduino default (or whether Marlin uses some servo library to set a specific frequency). These are things that could potentially be sources of trouble. However, the first step would be to get some measurable PWM output on a pin you can use, and then if the probe still doesn't work we could narrow it down from there.

Hope that helps! If you're still stuck, let me know and we'll try and figure this out.

wesk-ms, 2019-04-29:
Thanks for the reply.  No need to apologize, I assumed you were busy.

Thanks for confirming that not having these pins set up for PWM in PeripheralPins.c is the cause of the behavior I'm seeing.  I tried enabling them, to no avail, but I think I may have caused a timer conflict in doing so.  I noticed that in order to get add the pins I wanted, I was likely going to have to redefine a chain of other pins to ensure no sharing of timers.

I agree that the bltouch seems to be finicky and that getting Marlin to simply output a PWM signal when issuing a bltouch command is my next step.  Fortunately, I'm not using the serial pins on exp3, so that's where I'll go next.  Once I get that working, I'll figure out how to correctly set up the other pins in PeripheralPins.c.

Lastly, can you tell me where you're actually mapping the pin names (like PD_0, PC_5, etc...) to the physical pin numbers on the processor?  I thought that would be done in variant.h, but it's not.  I recall some code elsewhere that looked like it was doing the conversion, but I'm not sure.  Specifically, where does PD_15 get mapped to physical pin 62?

Thanks again for your help!  I was going nuts trying to get my DUE+RADDS board to perform well and work reliably, so I finally decided to scrap that and find a better 32bit board.  You designed the perfect solution IMO, and I think it should become the new 'standard' for 3D printer controllers.

chrissbarr, 2019-04-29:
No worries, happy to help.

The pin mapping is done in the variant.cpp file, starting from about line 25, here:

https://github.com/stm32duino/Arduino_Core_STM32/blob/master/variants/RUMBA32_F446VE/variant.cpp

It's not immediately obvious that that's what you're after, because the way the pins are defined in stm32duino you don't actually see the pin numbers. The pin numbers are implicit in the list - so the first pin is D0, the second is D1, and so on. 

The numbering is somewhat arbitrary, and there's _no correlation_ between the numbers assigned to the pins and their actual physical numbering - I simply chose to go from PA_0 in order. This may be something that's tripped you up - to my knowledge, the physical pin numbers are not used anywhere in the stm32duino core (or at least, not for this board). You can use either the port/pin definitions (i.e. PA0, PD15, etc), or the number mapped in the above file - which for PD15 is 63.

It would be nice to have the stm32duino pin numbers correspond to the physical package numbers, but I'm not sure if this is doable - specifically, I'm not sure if it's possible to have gaps in the pin numbers, as would be needed to allow for power and other non-gpio pins.

Hope that helps! Let me know if you need any more info.

DropbearNinja, 2019-11-27:
Hi @WK-MS did you get bltouch working on the rumba32? I'm having same issue

WK-MS, 2019-11-27:
No, I didn’t.  I think the issue was in the STM32 core.  PWM timer related.  The Rumba32 board is incredible, but because of that issue I moved to a different board and had no problems.  I’ve also moved away from Marlin and I’m using Klipper with much better results.  You might want to try MK4Duo firmware.  The owner of that code has recently ported it to STM32…specifically for the Rumba32 board.

Thanks,
Wes

DropbearNinja, 2019-12-05:
Hi @chrissbarr - could you please have a look at this? I've spent about 2 weeks trying literally everything.... I looked at the data sheet, pin PD15 is alt fn of TIM4, I created that in peripheralpins
{PD_15,  TIM4,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM4, 1, 0)},
And tried, well basically everything, even stuff that didn't make sense.
The rumba32 is a great board, but extra PWM functionality is a necessity, and at the moment non functioning....

DropbearNinja, 2019-12-05:
... 2 weeks, I post, and then 1 hour later, get it working...

DropbearNinja, 2019-12-05:
In 
> C:\Users\xxx\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\variants\RUMBA32_F446VE

change 
`#define TIMER_SERVO             TIM7`
to
`#define TIMER_SERVO             TIM4`
(reason: PD15 which is PWM2 is connected to TIM4 as per stm32f446 datasheet)

In:

> C:\Users\xxxxx\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\variants\RUMBA32_F446VE\PeripheralPins.c

change:
`//  {PB_9,  TIM4,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM4, 4, 0)},  // TIM4_CH4`
to:
`{PD_15,  TIM4,   STM_PIN_DATA_EXT(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF2_TIM4, 4, 0)},  // TIM4_CH4`

In Marlin in:
`..\Marlin-2.0.x\Marlin\src\pins\stm32\pins_RUMBA32.h`
add
`#define SERVO0_PIN	PD15`

That's it... that's all you need to do, and PWM works on PD15, and hence bltouch works fine

closed 2020-07-08 by chrissbarr