Aus3D/RUMBA32 #9: Error in BoM

leseaw, 2019-07-08:
Hello,

Point 6 in RUMBA32_v1_0E_BOM.xlsx
Name:
CAP CER 4.7UF 10V X7R 0603
PartNumer:
	TDK 
C1608X7R1C105K080AC
C Series 0603 1 uF 16 V ±10 % Tolerance X7R SMT Multilayer Ceramic Capacitor

i think Name is wrong or partnumer?

chrissbarr, 2019-07-08:
Hi @leseaw,

Yes, you're right, there's an issue with this line in the BOM. We noticed this recently but it hasn't been corrected yet. I don't believe we were able to find a part matching the specifications we gave in an 0603 package - in our current batch of boards we have used the following part:

C1608X7S1A475K080AC - CAP CER 4.7UF 10V X7S 0603 (Digikey [445-14258-1-ND](https://www.digikey.com/product-detail/en/tdk-corporation/C1608X7S1A475K080AC/445-14258-1-ND/3955924))

That matches the capacitance, voltage, and package, but is dropping from an X7R to X7S. This is just a small tolerance change and won't be a problem in this application, as the capacitance value is not critical.

We will likely be updating the BOM to this or a similar part soon.

leseaw, 2019-07-08:
Hi @chrissbarr ,

Are there any more changes which are still wrong in the BoM?

chrissbarr, 2019-07-08:
Not that I am aware of. Apart from that one exception, we are using the BOM as included without any changes.

leseaw, 2019-07-08:
okay thanks @chrissbarr 
And read your E-Mails :)

closed 2019-07-08 by leseaw

chrissbarr, 2019-07-08:
No worries @leseaw - just to be sure, you did get my reply last week?

leseaw, 2019-07-08:
@chrissbarr yes :) this i s why i write here. i answer you.

leseaw, 2019-07-09:
@chrissbarr my mails to you come back :( time out from your mailserver.