Aus3D/RUMBA32 #10: BoM and Parts D3 D4 D5 D15 component is SS24FL diode

leseaw, 2019-07-18:
Hi Chris,

i think thats wrong or?

D3, D4, D5, D15, the BOM component size is 5.4*1.8mm, while Gerber pad size is 3.7*1.2mm, not match.

chrissbarr, 2019-07-18:
Hi @leseaw,

From what I can see in the SS24FL [datasheet](https://www.onsemi.com/pub/Collateral/SS26FL-D.PDF), it measures a maximum of 3.90 x 1.95mm. Maybe double-check where the 5.40 x 1.80mm size you've mentioned comes from, as I couldn't find it.

The pads we have are a little smaller than the recommended footprint and don't extend beyond the diode's leads, but we haven't had any trouble with soldering these parts (reflow has always worked fine and I've assembled/reworked a number of boards by hand without noticing a problem here). We've used this same footprint & part since V1.1 of the RUMBA+ boards, and I don't believe we've ever seen a problem with these diodes - so I don't think you will run into any trouble there. 

I'll take a look at the pads during the next revision, but I'm hesitant to do a revision just for this as it's always worked without issue.

Hope that helps!

leseaw, 2019-07-19:
My mistake, ashes on my head :(

closed 2019-07-19 by leseaw