Aus3D/RUMBA32 #11: platformIO setup for Marlin

wap32, 2019-08-30:
Hello,

I just got myself a RUMBA32 board and am currently setting it up with Marlin.
The WiKi is helpful enough to guide through the Arduino IDE setup, and this works fine and I'm able to compile and upload without issues (so far).

However, I would like to get Marlin working with platformIO, and here I've been facing some issues.

Is there a known good platformio.ini env setup for Marlin that compiles properly with RUMBA32?

I think this would be a good addition to the WiKi documentation, as platformIO tends to be more straightforward to use once set up properly, and more portable as it does not depend on an existing properly configured Arduino IDE.

lei111, 2019-11-23:
I also encountered the same problem and hope to provide a configuration example. Thank you

wap32, 2019-12-10:
I forgot to update this issue, I ended up finding a working config for platformio, at least as of a couple of months ago (have not tested with latest Marlin build).

```
#
# RUMBA32
#
[env:rumba32_f446ve]
platform      = ststm32
framework     = arduino
board         = rumba32_f446ve
build_flags   = ${common.build_flags}
  -DSTM32F4xx
  -DARDUINO_RUMBA32_F446VE
  -DARDUINO_ARCH_STM32
  "-DBOARD_NAME=\"RUMBA32_F446VE\""
  -DSTM32F446xx
  -DUSBCON
  -DUSBD_VID=0x0483
  "-DUSB_MANUFACTURER=\"Unknown\""
  "-DUSB_PRODUCT=\"RUMBA32_F446VE\""
  -DHAL_PCD_MODULE_ENABLED
  -DUSBD_USE_CDC
  -DDISABLE_GENERIC_SERIALUSB
  -DHAL_UART_MODULE_ENABLED
  -Os
lib_deps      = ${common.lib_deps}
lib_ignore    = Adafruit NeoPixel
src_filter    = ${common.default_src_filter} +<src/HAL/HAL_STM32> +<src/HAL/HAL_STM32_F4_F7> -<src/HAL/HAL_STM32_F4_F7/STM32F7>
monitor_speed = 500000
upload_protocol = dfu
```

deruben900, 2020-01-21:
Have you been able to use the board with the latest Marlin version (2,0) with platformio? 
I am able to build, but not upload it firmware to my board.

capibara1, 2020-06-01:
I have not been able to compile Marlin 2.0.x (2.0.5.3 as of today) or bugfix-2.0.x at all. Whether is it with the Arduino IDE or via Platform IO. The platform.ini entry for the RUMBA32 that @wap32 provided seems to be the one to use (better than the default entry that comes with Marlin) but still doesn't produce a successful compile.

chrissbarr, 2020-06-02:
The PlatformIO config included with the current `bugfix-2.0.x` Marlin branch is working for me - I've put together some instructions here: https://github.com/Aus3D/RUMBA32/issues/22#issuecomment-637216106

chrissbarr, 2020-07-07:
I've updated the [https://github.com/Aus3D/RUMBA32/wiki](wiki) with instructions for using PlatformIO.

closed 2020-07-07 by chrissbarr