Aus3D/RUMBA32 #13: TMC220x on-board UART compability

Blaster1920, 2019-09-29:
Hi,
by having a look at the design I noticed that it is missing an important feature to correctly drive the TMC2208 and TMC2209 drivers directly from the MCU. In fact, it looks like you omitted the possibility of choosing between SPI and UART mode: is there a particular reason why you preferred going this way?
Thank you in advance for your time.

chrissbarr, 2019-09-30:
Hi @Blaster1920,

Short answer: supporting the UART drivers didn't seem feasible within the constraints of this board, but if you can see a way to make it work let me know and we'll take it into consideration.

My understanding is that the UART-based TMC drivers require two communication pins (TX and RX) per driver. Please correct me if this is incorrect. The SPI-based TMC drivers require three pins shared among all drivers (MISO, MOSI, SCK) and 1 pin for each driver (CS). On the current design, there are not enough spare GPIO to support adding an extra pin to each driver (also it would be tough to route an extra signal to each driver socket).

If there's a way that the UART drivers can work with just a simple jumper using the signals that are already routed to each socket, then it's definitely something I'd be interested in exploring. At the moment, each stepper driver has the following pins/signals:
- STEP
- DIRECTION
- ENABLE
- CHIP SELECT
- MISO (shared)
- MOSI (shared)
- SCK (shared)

So, we don't have two pins per driver (outside the usual STEP/DIR/EN) to act as RX & TX. Is it possible to control the TMC2208 with only TX (i.e. no feedback)? If so, it might be possible to use the CHIP SELECT signal as a soft UART TX for that purpose. I'm not familiar enough with the state of TMC220X firmware support to say.

Let me know what you think, if you have a way of making it work I'd love to get it included in the design.

Blaster1920, 2019-09-30:
Hi @chrissbarr,
thank you for the prompt and detailed response.
I've been looking for possible solutions, but so far nothing seems to be what we need, even though there may be a couple of possibilities:

- [https://github.com/MarlinFirmware/Marlin/issues/14208](url) a bug report that may be fixed by now for using Tx only mode.

- In the STM32F446VE datasheet 6 USARTs are being mentioned, which ones would you have free if you add an onboard microsd reader? Would it be possible to share the same pins for both usb and microsd reader by enabling usb only when the microsd is not inserted?

- [https://www.instructables.com/id/UART-This-Serial-Control-of-Stepper-Motors-With-th/](url) an extra link that may be handy.

- Last but not the least: what if you enable the UART to TMC22xx on-board only for 4 drivers? That's usually the typical need when tmc2208 are used. (x y z e0)

Let me know.

sverd, 2020-08-03:
Hi,
I noticed that in newer revisions the on-board UART seems to have been fixed/added. I could however not completely figure out how this was done. The thing is that I have a 1.0 version of the board and I'm wondering whether it is possible to "fix" that board to get UART working with the stepper drivers. Is it possible to get UART working with possibly a few  small modifications? (mainly asking because I saw the fix to get SPI communication with the stepper drivers working which is very doable)