Aus3D/RUMBA32 #14: Exp3 pinout

MagoKimbra, 2019-10-11:
Hi, I'm here again to ask you something, I think it's another mistake I think of drawing.
The pins in exp3 5 and 6 PWM 1 and PWM2 are on processor PD14 and PD15, so pins 13 and 14 what are they? PD12 and PD13? On the drawing they are called PD14 PD15 ...
If they are PD12 and PD13, you know if they can be used as interrupts, STM32 can use all pins as interrupts, if you know it, otherwise I try ...

chrissbarr, 2019-10-13:
Hi @MagoKimbra,

You're right again! Yes, PD14 and PD15 are misnamed, they should be PD12 and PD13.

I've updated the schematic and images to show this correctly now. 

All pins can be used as interrupts, but I believe only one of each pin number (i.e. can't use both PA**0** and PB**0** as interrupt - only PA0 or PB0 or PC0 etc...). So there are 16 pin-change interrupts available (Px0 to Px15). I included a table in the schematic showing one way the pins can be used without conflicting:

[image]

So, if you want to use interrupts for X-MIN and X-MAX (PB12 and PB13) then there are no interrupts available for PD12 and PD13. To use interrupts on PD12 and PD13, there can be no interrupts on any other Px12 and Px13 pins.

MagoKimbra, 2019-10-13:
Hi, please check carefully, because yesterday I tried pin by pin and the correspondence is not that.
What I found was this:
[image]

chrissbarr, 2019-10-13:
Hi @MagoKimbra,

It's easy to get mixed up here, so I'll put together all the info I have and we'll see if we can get this figured out.

Here's an image showing the pins in the schematic and on the PCB:

[image]

Also included is the pin definition in the Arduino_Core_STM32 variant.h file, which sets the Arduino pin numbers for the STM32 pins.

The numbers in your image - are they the Arduino pin numbers? The Arduino pin numbers should be:

PD12 (previously PD14 as in your image) - 60
PD13 (previously PD15 as in your image) - 61
PWM_1 (PD14) - 62
PWM_2 (PD15) - 63

The first three are the same as in your image, but PWM_2 is different. Are you sure that PWM_2 corresponds to 59? I can't find this in my checking, but I am prepared to be wrong.

MagoKimbra, 2019-10-13:
Yes, I was wrong on 59, I tried only 60, 61 and 62 so I had one pin left and I wrote 59, in fact it's 63, perfect now everything exp3 matches me.

MagoKimbra, 2019-10-13:
https://www.facebook.com/MK4duo/videos/857002628047547/
But you must write to group for see the video...

chrissbarr, 2020-07-07:
I'm going to close this issue out now, but for future reference the pinout of the EXP3 header is:

[image]

I've added a pinout image to the main readme.

closed 2020-07-07 by chrissbarr