Aus3D/RUMBA32 #15: Where Rumba32 is manufactured

MagoKimbra, 2019-12-12:
Hi, there is a discussion between us in Italy, some people have found a German site where they sell Rumba32 for 50 € but made in China.
What do you sell on the Aus3d website made by you in Australia or is that made in China?
If it's made in China, tell me why it costs almost three times as much?
I for my choice do not buy any product made in China. So knowing that your card is made in Australia makes me happy, but if it is made in China I would be deeply saddened.
Thanks for any replies.

BoHoehne, 2019-12-17:
Hello,
I just bought one of the "German Site" 50 EUR boards. 
On the back side it says: 

Designed in Australia, made in China.

Have a good day.

MagoKimbra, 2019-12-17:
Yes, I know, I wanted to know the one bought in Australia on the AUS3D website. I hope it's made in Australia and not in China.
I believe that Chinese quality is very low, so I don't buy anything that is made in China.

celulu08, 2019-12-21:
Hello, I just received the rumba32 from China, it is well marked AUS3d, designed in Australia, made in China.

Celcius1, 2020-10-04:
I know Chris personally, all I can say is the boards are made in china, but the ones Chris sell are vastly different to the cheap copies, and to show the blatant copying, they don't even remove his business name in some cases, a full copy, but the difference is when Chris gets the assembled boards, he does some minor finishing touches to the microcontroller on the board, to give the boards that extra bit of polish.  So its safe to say, the Aus3D ones are far better as I am aware Chris tests each one before they go out, so they function as intended.