Aus3D/RUMBA32 #16: Flash without button press

lorenzing, 2019-12-19:
Hi, 
I ask you if I can flash the card without pressing the boot and reset keys, since the card is in an awkward position in the printer. Thank you very much for helping.

Lorenzo

robustini, 2019-12-19:
Great request, if the baord is installed inside the printer in a difficult place it is absurd to have to press the buttons on the pcb.
On SKR have created a bootloader that solves this problem, the ST CPUs are usually sent in DFU mode with a usb command, because here is the question of the buttons?
Uncomfortable!

robustini, 2019-12-19:
https://github.com/Arksine/LPC17xx-DFU-Bootloader/releases/tag/0.4.1?fbclid=IwAR2tZPzka_RSKFu7V2b9JSu-P-opkz3r-4v0mxCES-hfVaRA15Fyy6hMPMI

Winky1st, 2019-12-21:
MagoKimbra with his Mk4duo 4.4.0 has solved it. Command is M9999, but are needed  github libraries for arduino ide. With stock one won’t work yet until ide libs will be upgraded. 

https://marlinkimbra.it/configurator/v4_4_0/firmware_configurator.php