Aus3D/RUMBA32 #17: Stepper sockets in Rev E still defect

repetier, 2020-01-08:
Hi @chrissbarr ,
I purchased a RUMBA32 Rev E to add support for your board to Repetier-Firmware V2 (https://github.com/repetier/Repetier-Firmware/tree/dev2) and it is working already very good.

I read that up to Rev D there is a problem with SPI MOSI pin connected to wrong pin header, so was lucky to have a "fixed" board. But after searching for hours I detected that reading the driver does not work due to same pin switch. So not sure if it was meant to be fixed or if that was Rev E in old version or if it is different producer since I ordered it in Germany, but it contains your domain on backside.

Beside that it would be nice if you would add a reference to the additional firmware possible with that board.

leseaw, 2020-01-08:
Rev E has this error too :(

chrissbarr, 2020-06-02:
Hi @repetier, @leseaw.

Sorry there was no reply on this earlier. I have not been getting notifications from this repo, and am going through at the moment trying to get caught up.

To confirm, the SPI defect is present in v1.0E. I have put together a central FAQ issue on this here, for future reference: https://github.com/Aus3D/RUMBA32/issues/23

Really cool to hear that the board is being supported within Repetier-Firmware V2! I will give it a go when I receive the next batch of boards.

chrissbarr, 2020-06-02:
I have added a link to Repetier-Firmware in the readme:

[image]

Let me know if you would prefer different wording or a different URL linked to.

closed 2020-06-02 by chrissbarr