Aus3D/RUMBA32 #18: MKS TFT32_L V4.0 compatibility

rubenoosterwijk, 2020-01-10:
Is the Rumba32 board not compatible with the Makerbase Touchscreens LCDs?
The Rumba+ board did have a AUX1 port.
How can i fix this?

kai658, 2020-02-26:
MKS RUMBA32, which I bought at maker base, is supported

chrissbarr, 2020-07-08:
Hi @rubenoosterwijk,

I'm hoping you've solved this by now, but will post an answer in case anyone comes across this for future reference.

The RUMBA32 board has the same expansion headers that RUMBA+ had -  EXP1, EXP2 and EXP3.

I'm not familiar with an AUX1 port on RUMBA+ - perhaps this is something that another manufacturer added to their version?

From what I can see, the MKS TFT displays need 5V, GND, UART TX and UART RX. You can access all of these on EXP3 - refer to the image below:

[image]

I'm not sure what type of cable the display comes with. You'd need to find a way to connect the correct pins.

closed 2020-07-08 by chrissbarr