Aus3D/RUMBA32 #20: BL Touch

danielbors72, 2020-03-12:
how do i connect bltouch?

phongshader, 2020-04-22:
I would like to know as well.

leseaw, 2020-04-22:
https://github.com/Aus3D/RUMBA32/issues/8

read this :)

Celcius1, 2020-10-04:
The Servo Header should do it, but if you have the version (which I have) that does not have a servo header, make the following changes to connect a BL Touch

in the pins_RUMBA32_common.h

change Z_MAX_PIN from PD10 to PA4 (just select a random unused pin)

change SERVO0_PIN from PA15 to PD10
this utilises the Z_MAX_PIN for the BL Touch like that has been used on some boards but be mindful to run the 5v for the BL Touch to a 5V power pin, as the endstop pins are configured for 3.3v by default

and do the usual changes to get the BL Touch working.