Aus3D/RUMBA32 #22: Instructions for RUMBA32 Marlin compile are obsolete

capibara1, 2020-06-01:
Hi Chris,

I created this issue based on the comments in issue #21 and issue #11 

The instructions to get the board up and running with Marlin are obsolete. Here are the specifics that need to be updated:

Based on the instructions here:
 [https://github.com/Aus3D/RUMBA32/wiki/Getting-RUMBA32-Up-&-Running](https://github.com/Aus3D/RUMBA32/wiki/Getting-RUMBA32-Up-&-Running)

1. Install Arduino IDE & STM32duino Core section

Following the instructions for stm32duino used to work fine until Nov 2019. Now, whatever version of STM32 Cores that is used (access in the Arduino IDE via Tools->Board->Board Manager) from 1.6.0 to the latest 1.9.0 do not produce a successful compile. Instructions need to be updated to indicate what version of these libraries compile with specific versions of Marlin

2. The Marlin board definition instructions no longer apply to either Marlin 2.0.x or Marlin-bugfix-2.0.x. There are two versions of the RUMBA32:

Change  #define MOTHERBOARD BOARD_RAMPS_14_EFB to #define MOTHERBOARD BOARD_RUMBA32

It should now be: 

Change  #define MOTHERBOARD BOARD_RAMPS_14_EFB to **#define MOTHERBOARD BOARD_RUMBA32_AUS3D**

Even with the changes in the instructions, I think using the Arduino IDE makes compiling and keeping track of all the library versions much more difficult. There should be specific instructions to set up the Platform IO environment for RUMBA32, but that's another issue.

Thanks!

chrissbarr, 2020-06-01:
Thanks @capibara1. I'll have a look at updating this shortly.

chrissbarr, 2020-06-02:
Hi @capibara1,

I haven't tried the Arduino IDE yet, as the last time I did I ran into a few problems, but I've confirmed that the following PlatformIO setup works for me:

1. Using VSCode with PlatformIO installed (other IDEs are probably similar, but I have not tried).
2. Download/clone Marlin 2.0.x (see below for notes on version) and open it in VSCode.
3. Edit the configuration.h file with the minimum necessary changes:

```
107: #define SERIAL_PORT -1

130: #ifndef MOTHERBOARD
131:   #define MOTHERBOARD BOARD_RUMBA32_AUS3D
132: #endif

409: #define TEMP_SENSOR_0 998 // not necessary, but so I could test without a thermistor
```

4. Locate the `env:rumba32_f446ve` folder in the `Project Tasks` window, and expand it:

[image]

5. Click on `Build` in the `env:rumba32_f446ve` folder. You should see a few success messages once the process completes:

[image]

[image]

6. Connect RUMBA32 to the USB port. Reset the RUMBA32 board by holding BOOT0, pressing RESET, then releasing BOOT0. The LED should blink once when you do this. Device manager should show the board as so:

[image]

7. Click on `Upload` in the `env:rumba32_f446ve` folder. Again, you should see a few success messages once the process completes:

[image]

Here's the full output I get when building/uploading:

<details> 
  <summary>Full Output (Click to Expand)</summary>

```
> Executing task in folder Marlin: C:\Users\Chris\.platformio\penv\Scripts\platformio.exe run --target upload --environment rumba32_f446ve <

Processing rumba32_f446ve (platform: ststm32; board: rumba32_f446ve; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/rumba32_f446ve.html
PLATFORM: ST STM32 6.1.0 > 3D Printer control board
HARDWARE: STM32F446RET6 180MHz, 128KB RAM, 512KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 4.10800.200207 (1.8.0)
 - tool-dfuutil 1.9.200310
 - tool-openocd 2.1000.190707 (10.0)
 - tool-stm32duino 1.0.2
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 20 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <LiquidCrystal> 1.5.0
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.6.2
|   |-- <SoftwareSerial> 1.0
|   |-- <SPI> 1.0
|-- <U8glib-HAL> 0.4.1
|   |-- <Wire> 1.0
|   |-- <SPI> 1.0
[... 147 lines cut ...]
FYSETC_S6                 IGNORED
STM32F407VE_black         IGNORED
BIGTREE_SKR_PRO           IGNORED
BIGTREE_GTR_V1_0          IGNORED
BIGTREE_BTT002            IGNORED
teensy31                  IGNORED
teensy35                  IGNORED
esp32                     IGNORED
linux_native              IGNORED
SAMD51_grandcentral_m4    IGNORED
rumba32_f446ve            SUCCESS   00:00:25.503
rumba32_mks               IGNORED
include_tree              IGNORED
================================================================================================== 1 succeeded in 00:00:25.503 ==================================================================================================
```
</details>

8. Once the upload is complete, the status LED on the board should begin to flash quickly (~ 4 times per second). You should be able to communicate with the board using your printer host software.

### Marlin versions
I tested with the following versions of Marlin:
- The latest in the `bugfix-2.0.x` branch, which was https://github.com/MarlinFirmware/Marlin/commit/53cb2963dc3091828e6aedc2deec8a123b3634eb. This worked for me and seemed to run correctly (I did not test every feature, obviously, but it boots and communicates).
- The latest in the more stable `2.0.x` branch, which was https://github.com/MarlinFirmware/Marlin/commit/e4d8336175ee09b28d23220f446d07451ca3de43. This did not work for me - it compiles but I get no LED heartbeat and no serial communications. Possibly this is out of date with the current STM32 core.

Let me know if you give the above a go, and if it works for you or if you run into trouble. If it does the trick I'll look at updating the instructions to include these steps.

capibara1, 2020-06-02:
See comments here: [Issue #24] (https://github.com/Aus3D/RUMBA32/issues/24#issuecomment-637230802)

chrissbarr, 2020-07-07:
I've updated the [wiki](https://github.com/Aus3D/RUMBA32/wiki) with instructions using PlatformIO, based on our conversations here and some changes that have been pulled into Marlin since. These instructions are currently up to date - hopefully that stays the case for a while!

closed 2020-07-07 by chrissbarr