Aus3D/RUMBA32 #23: [FAQ] SPI Stepper Driver Defect

chrissbarr, 2020-06-02:
The purpose of this issue is to put info on this problem in one place for easy reference.

As has been discussed in several issues in this repository and others, there is a design defect with the SPI pinout in the stepper driver sockets that prevents bidirectional SPI communication with stepper drivers from working properly. Because of this, SPI support for stepper drivers in Marlin and other firmware does not work correctly.

This issue is present on all board versions from v1.0A to v1.0E. For reference, v1.0E is the only revision sold by Aus3D (the earlier revisions were only used for internal development). I have seen some vendors selling clones based on v1.0D, so I am trying to cover my bases by describing all models, whether they have been sold by Aus3D or not.

At the moment, all clone boards from other vendors seem to be based on either v1.0D or v1.0E, and this issue is present in all those boards that I have seen. This includes MKS's modified RUMBA32 board (currently v1.0), which looks to be derived from v1.0E.

If you want to get SPI-based drivers working, the following instructions describe one method:

[image]

MKS describes a slightly different method in their FAQ [here](https://github.com/makerbase-mks/MKS-RUMBA32/wiki/About-RUMBA32-TMC-SPI-%22CONNECTION-ERROR%22-FAQ), in which the SLP and RST pins are soldered together. This should get the SPI signals to the correct pins, but it also puts one of the signals on another pin - whether or not all models of stepper driver will work correctly with this is unclear to me.

chrissbarr, 2020-06-02:
### Other discussions on this issue:
First discussed in this issue: https://github.com/Aus3D/RUMBA32/issues/12#issuecomment-536172666
Discussed in MKS RUMBA32 issues here: https://github.com/makerbase-mks/MKS-RUMBA32/issues/2

studiodyne, 2021-01-28:
Hi , may be i can suggest something .... You have to redraw the new board , may you can add some features ...
I give you all my ideas , may be some will interest you , and after reading this , you can delete , if you want to keep secret these ideas
1-) Create a new expanion port with   all ' step + dir ' pins , that will make 12 gpios available when using SPI drivers only ,  this new port give infinite possibilities , add 12 motors , or servos or alarm pins or robotics pins or anything else
2-) Create a 6v power for servos , because , the 5v works only on very short cables , but when using long cables , the servos don't want to work because of too low voltage
3-) Add 2 jumpers , for the EEPROM for users that want more PROGMEM , or prefer EEPROM , or ' like  me ' that want the 2 free pins of the i2c bus
4-) create a three pins dedicated bltouch port , i know it's risked if one day a new firmware breaks something in the timers and pins ,that breaks bltouch pwm , but for the moment it's works fine on pb8/pb9/pd15 and others
5-) add MKS tft connector on the same sw uart pins
6-) big connector for heaters , same as power supply because ' super volcano 80w and i'm sure , very soon a new ultra volcano 150w ' , the mosfet can drive , but connectors .... not sure that have the size of the cable
7-) another type of buttons , because , resetting and bootloading is not easy 

Thks for your job