Aus3D/RUMBA32 #25: jumpers to set for DRV8825 drivers

phongshader, 2020-06-06:
I don't see any documentation on how to set the jumpers for the drivers. If I want to set up x and y drivers to 16 micro steps and z to 32 micro steps. SLP MS3 MS2 MS1 or maybe I don't understand how this works. Documentation?

3dlpm, 2020-06-07:
@phongshader I actually recommend that you don't use DRV8825 drivers for your 3d printer at all. They are bad for reasons explained:
1. Here: [https://forum.pololu.com/t/poor-drv8825-microstepping-performance/15266](https://forum.pololu.com/t/poor-drv8825-microstepping-performance/15266)

2. And here (look at the links in the "External notes/references" section) : [https://reprap.org/wiki/DRV8825](https://reprap.org/wiki/DRV8825)

A4988 stepper drivers are proven to be better to use for 3D printers than the DRV8825.
Another alternative, if you want silent operation and to be able to control currents and other parameters via the built-in RUMBA32 SPI interface, you want to use TMC2130 or TMC5160 drivers which are SPI controllable. I know the RUMBA32 definitely supports TMC2130s but not sure if it supports the higher currents driven by the TMC5160.

In any case, for DRV8825 the jumper settings are in the picture below. They are in the RUMBA32 schematic in this GitHub page and you can also do a search on DRV8225 pinout.

[image]

phongshader, 2020-06-08:
Thank you, I will be changing over to TMC5160 drivers. I'm upgrading from a rumba to rumba32 and I want to tackle one issue at a time, switch the board with the drivers I have that I know will work with this board, get it running , then switch the drivers that have their own set of problems with this board

closed 2020-06-09 by phongshader