Aus3D/RUMBA32 #26: Fan PWM Troubleshooting / Timer Investigation

chrissbarr, 2020-06-06:
Collecting information for troubleshooting why PWM on FAN (PC9) isn't outputting the expected values when running Marlin bugfix-2.0.x.

Suspect there is a conflict somewhere / the same timer is being used in multiple places or something to that effect.

## Timers in use
Checking everywhere timers are configured/selected.
### STM32 Arduino Core
[RUMBA32_F446VE/variant.h](https://github.com/stm32duino/Arduino_Core_STM32/blob/master/variants/RUMBA32_F446VE/variant.h):
```
// Timer Definitions
// Use TIM6/TIM7 when possible as servo and tone don't need GPIO output pin
#define TIMER_TONE              TIM6
#define TIMER_SERVO             TIM7
```
[SoftwareSerial/src/SoftwareSerial.cpp](https://github.com/stm32duino/Arduino_Core_STM32/blob/master/libraries/SoftwareSerial/src/SoftwareSerial.cpp)
```
// It's best to define TIMER_SERIAL in variant.h. If not defined, we choose one here
// The order is based on (lack of) features and compare channels, we choose the simplest available
// because we only need an update interrupt
#if !defined(TIMER_SERIAL)
  #if defined (TIM18_BASE)
    #define TIMER_SERIAL TIM18
  #elif defined (TIM7_BASE)
    #define TIMER_SERIAL TIM7
  #elif defined (TIM6_BASE)
    #define TIMER_SERIAL TIM6
  #elif defined (TIM22_BASE)
    #define TIMER_SERIAL TIM22
```
### Marlin
[STM32/timers.cpp](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/HAL/STM32/timers.cpp):
```
#elif defined(STM32F401xC) || defined(STM32F401xE)
  #define HAL_TIMER_RATE (F_CPU / 2)
  #define MCU_STEP_TIMER  9
  #define MCU_TEMP_TIMER 10
#elif defined(STM32F4xx) || defined(STM32F7xx)
  #define HAL_TIMER_RATE (F_CPU / 2)
  #define MCU_STEP_TIMER  6           // STM32F401 has no TIM6, TIM7, or TIM8
  #define MCU_TEMP_TIMER 14           // TIM7 is consumed by Software Serial if used.
#endif
```

## Timers Reference
[image]

phongshader, 2020-06-08:
I don't know if the issue I'm having is connected to this but it sounds like it is. If I run a m106 command FAN0 does not turn on however FAN0 led does light up. The voltage at the fan terminal after a m106 s255 is 1.39 and s128 is .93, s0 is .10

chrissbarr, 2020-06-08:
Hi @phongshader,

The problem you're describing is the reason I started looking into this. However, I've found the fan issue is not related to any of the above, but is an issue with the PWM frequency+resolution in 1.7.0 of the Arduino Core (which is the version ProjectIO + Marlin currently uses). When ProjectIO and Marlin update to use v1.9 this issue should be fixed.

In the meantime, the best way to get fans running correctly is to use Marlin's `#define FAN_SOFT_PWM` feature - all fans should work when that is enabled.