Aus3D/RUMBA32 #27: 12v pwr select

phongshader, 2020-06-06:
I don't see any documentation on how to set the jumpers for 12v power. I did see this
[image]
however it is not clear to me how these should be jumped for 12v power,  vin > vout or vout > 12v?

3dlpm, 2020-06-06:
Hopefully this picture will clear it up. For 12V power, 12V to Vout for PSU power Vin to Vout.

[image]

chrissbarr, 2020-06-07:
Thanks for the image @3dlpm, your explanation is correct.

phongshader, 2020-06-07:
Thank you for the explanation. What does each set of jumpers do, why is the left set vout > 12v, middle vin > vout, right vin > 12v? I just want to understand what I'm setting up and why.

chrissbarr, 2020-06-07:
Each jumper sets the voltage for the output it's next to. HE2, FAN0 and FAN1 can be powered either directly from the PSU (so if you have a 12V PSU, it's 12V, if you have a 24V PSU, it's 24V) or from RUMBA's integrated 12V regulator (which will output 12V even if you are using a 24V PSU). 

If you are running a 12V PSU, you should just set all to VIN. If you are running a 24V PSU but have 12V fans, you may want the 12V option selected. 

The top pin of each of the three pin jumpers is the regulated 12V. The bottom pin is the raw VIN from the PSU. The middle pin is what goes to the output. The jumper connects the output to either of these depending on which position it's in.

In the picture:
The left-most jumper is setting the HE2 output to be powered from the regulated 12V.
The middle jumper is setting the FAN0 output to be powered from the VIN (the PSU).
The right-most jumper is setting the FAN1 output to be powered from the regulated 12V.

Let me know if more explanation is required.

phongshader, 2020-06-07:
That's exactly what I was looking for. I do have a 12v PSU and all the jumpers are set to VIN, so no change is needed. Thank You!

closed 2020-06-09 by phongshader