Aus3D/RUMBA32 #28: dfu mode device not recognized

phongshader, 2020-06-09:
mac:
FS mode
[image]

dfu mode
[image]

Win 10
VCP_V1.5.0_Setup_W8_x64_64bits.exe installed, it recognized the board worked once to get firmware onto board but I can no longer update the firmware.
FS mode:
[image]

dfu mode
[image]

no STM32 bootloader.
Is there a way to load firmware without dfu?

platformio:
`Uploading .pio/build/rumba32_mks/firmware.bin
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Match vendor ID from file: 0483
Match product ID from file: df11
dfu-util: No DFU capable USB device available
*** [upload] Error 74`

chrissbarr, 2020-06-09:
Hi @phongshader,

Hmm, in the images the board is recognised and connected, but not in DFU mode. At the moment DFU mode has to be entered manually by pressing the buttons on the board - are you doing this? Refer to the instructions (from [here](https://github.com/Aus3D/RUMBA32/wiki/Uploading-Sketch#method-1-upload-via-usb-dfu)):

1. Press and hold RUMBA32 BOOT button.
2. Press and release RUMBA32 RESET button.
3. Release RUMBA32 BOOT button.
4. RUMBA32 is now in hardware bootloader mode and will accept firmware upload via USB DFU.

The orange LED on the board should flash once when you do the above (approximately after RESET is released). This indicates the board is in DFU mode and has a working USB connection. You should be able to see the device in device manager as so:

[image]

I'm not sure if you've seen the troubleshooting steps here: https://github.com/Aus3D/RUMBA32/wiki/Uploading-Sketch#troubleshooting

That CDC FS mode comes up shows that the USB connection is working correctly and the hardware is functioning correctly. CDC FS mode is the serial mode that the board enters when running user code. USB DFU is a different mode that is controlled by the hardware bootloader - the good news is that the DFU bootloader is built in and can not be erased/overwritten/removed. I would make sure that you are pressing the buttons firmly, in the correct order, and watch for the orange LED to make sure that the board is entering DFU mode.

phongshader, 2020-06-09:
I did indeed press boot and reset to get into dfu mode. The board does connect until I hold the boot button and press and release the reset button then it disappears from the device manager. I will check to see if the led, led1?,  flashes once when the reset button is released. 
The only way to load firmware, other than dfu, is with a usb to serial adapter?

chrissbarr, 2020-06-09:
Hi @phongshader,

The behaviour you are describing sounds fairly normal. The device should disappear from device manager when you press RESET, as resetting will disconnect it. It should reappear within a second or two of RESET being released, but may be under a different heading in device manager - for me, USB DFU ("STM32 BOOTLOADER") shows up under "Universal Serial Bus devices", whereas when it is in normal mode it shows up under "Ports (COM & LPT)". 

LED1 is the correct place to be looking. For me, it flashes for about 1/4 of a second after releasing RESET. I believe the LED lights when starting the bootloader, and turns off when the USB connection with the host computer is opened correctly. The exact timing _might_ vary slightly depending on the host PC / OS (I'm unsure if there can be differences in how quickly devices are enumerated etc.), but it should be a short flash that last some part of a second. If you try to enter the bootloader without the USB cable connected, the LED will stay lit for some time (> 60s) before it times out and stops trying to open the connection. If you see the LED staying lit that would indicate some connection problem - but the fact that it shows up when in CDC mode makes me think the connection is physically good.

There's no harm in holding BOOT down longer than needed, and there's no real timing requirement - as long as BOOT is held down when RESET is released, it should enter bootloader mode.

### Upload Methods
There are a few methods, but the USB DFU method should be the simplest and it's the only one I use regularly. It's also the only one that doesn't require any extra hardware. Upload methods currently supported in the Arduino IDE are:

- STM32CubeProgrammer (SWD)
  - Serial Wire Debug (SWD) is the dedicated programming/debugging interface for the MCU and can be used for debugging purposes as well as uploading etc. 
  - It requires a dedicated programmer such as the ST-Link V1/V2/V3 or similar. Adafruit carry one here: https://www.adafruit.com/product/2548
  - There are generic models on Aliexpress etc for a few dollars. I bought two at one stage and had one of them not work, so it may be a bit of a buyer-beware thing.
- STM32CubeProgrammer (Serial)
  - Program the MCU using the integrated bootloader via the serial U(S)ART interface. You need a 3.3V-compatible USB-Serial adaptor to do this, to be wired as shown in the wiki image:
[image]
- STM32CubeProgrammer (DFU)
  - Program the MCU using the integrated bootloader via the USB DFU interface.

I have used SWD programmers on other boards, but have not had to use them on RUMBA32. I have only used the Serial upload method once on these boards, to confirm I had the instructions correct when I was writing that wiki page. Both the USB DFU and Serial methods require the board to be in bootloader mode, so if you can make Serial work then DFU should work anyway.

phongshader, 2020-06-09:
I pulled out an ooold win7 netbook and it recognized the board in dfu mode. I was able to load firmware...it took several hours to compile. After that I was able to connect using a newer laptop. I don't know what happened but it doesn't seem to be a board issue

closed 2020-06-09 by phongshader