Aus3D/RUMBA32 #36: build failed with Arduino

19504643, 2020-08-24:
I'm using marlin bugfix-2.0.x branch, following the setup https://github.com/Aus3D/RUMBA32/wiki/Getting-RUMBA32-Up-&-Running#arduino-setup
first I got the board config error:
[image]
I changed `BOARD_RUMBA32` to `BOARD_RUMBA32_V1_0`

then I got errors: 

> Arduino: 1.8.13 (Windows 10), Board: "3D printer boards, RUMBA32, STM32CubeProgrammer (DFU), Enabled (generic 'Serial'), CDC (no generic 'Serial'), Low/Full Speed, Smallest (-Os default), Newlib Nano (default)"
> 
> sketch\src\HAL\STM32\timers.cpp: In function 'void HAL_timer_enable_interrupt(uint8_t)':
> 
> sketch\src\HAL\STM32\timers.cpp:178:62: error: no matching function for call to 'HardwareTimer::attachInterrupt(void (&)(HardwareTimer*))'
> 
>   178 |       timer_instance[timer_num]->attachInterrupt(Step_Handler);
> 
>       |                                                              ^
> 
> In file included from C:\Users\admin\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/stm32/analog.h:45,
> 
>                  from C:\Users\admin\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/board.h:8,
> 
>                  from C:\Users\admin\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/wiring.h:41,
> 
>                  from C:\Users\admin\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/Arduino.h:36,
> 
>                  from c:\users\admin\appdata\local\temp\arduino_build_271463\sketch\src\hal\shared\marduino.h:36,
> 
>                  from c:\users\admin\appdata\local\temp\arduino_build_271463\sketch\src\hal\stm32\hal.h:28,
> 
>                  from c:\users\admin\appdata\local\temp\arduino_build_271463\sketch\src\hal\hal.h:26,
> 
>                  from c:\users\admin\appdata\local\temp\arduino_build_271463\sketch\src\inc\marlinconfig.h:30,
> 
>                  from sketch\src\HAL\STM32\timers.cpp:24:
> 
> C:\Users\admin\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/HardwareTimer.h:134:10: note: candidate: 'void HardwareTimer::attachInterrupt(callback_function_t)'
> 
>   134 |     void attachInterrupt(callback_function_t callback); // Attach interrupt callback which will be called upon update event (timer rollover)
> 
>       |          ^~~~~~~~~~~~~~~
> 
> C:\Users\admin\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/HardwareTimer.h:134:46: note:   no known conversion for argument 1 from 'void(HardwareTimer*)' to 'callback_function_t' {aka 'std::function<void()>'}
> 
>   134 |     void attachInterrupt(callback_function_t callback); // Attach interrupt callback which will be called upon update event (timer rollover)
> 
>       |                          ~~~~~~~~~~~~~~~~~~~~^~~~~~~~
> 
> C:\Users\admin\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/HardwareTimer.h:138:10: note: candidate: 'void HardwareTimer::attachInterrupt(uint32_t, callback_function_t)'
> 
>   138 |     void attachInterrupt(uint32_t channel, callback_function_t callback); // Attach interrupt callback which will be called upon compare match event of specified channel
> 
>       |          ^~~~~~~~~~~~~~~
> 
> C:\Users\admin\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/HardwareTimer.h:138:10: note:   candidate expects 2 arguments, 1 provided
> 
> sketch\src\HAL\STM32\timers.cpp:181:62: error: no matching function for call to 'HardwareTimer::attachInterrupt(void (&)(HardwareTimer*))'
> 
>   181 |       timer_instance[timer_num]->attachInterrupt(Temp_Handler);
> 
>       |                                                              ^
> 
> In file included from C:\Users\admin\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/stm32/analog.h:45,
> 
>                  from C:\Users\admin\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/board.h:8,
> 
>                  from C:\Users\admin\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/wiring.h:41,
> 
>                  from C:\Users\admin\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/Arduino.h:36,
> 
>                  from c:\users\admin\appdata\local\temp\arduino_build_271463\sketch\src\hal\shared\marduino.h:36,
> 
>                  from c:\users\admin\appdata\local\temp\arduino_build_271463\sketch\src\hal\stm32\hal.h:28,
> 
>                  from c:\users\admin\appdata\local\temp\arduino_build_271463\sketch\src\hal\hal.h:26,
> 
>                  from c:\users\admin\appdata\local\temp\arduino_build_271463\sketch\src\inc\marlinconfig.h:30,
> 
>                  from sketch\src\HAL\STM32\timers.cpp:24:
> 
> C:\Users\admin\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/HardwareTimer.h:134:10: note: candidate: 'void HardwareTimer::attachInterrupt(callback_function_t)'
> 
>   134 |     void attachInterrupt(callback_function_t callback); // Attach interrupt callback which will be called upon update event (timer rollover)
> 
>       |          ^~~~~~~~~~~~~~~
> 
> C:\Users\admin\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/HardwareTimer.h:134:46: note:   no known conversion for argument 1 from 'void(HardwareTimer*)' to 'callback_function_t' {aka 'std::function<void()>'}
> 
>   134 |     void attachInterrupt(callback_function_t callback); // Attach interrupt callback which will be called upon update event (timer rollover)
> 
>       |                          ~~~~~~~~~~~~~~~~~~~~^~~~~~~~
> 
> C:\Users\admin\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/HardwareTimer.h:138:10: note: candidate: 'void HardwareTimer::attachInterrupt(uint32_t, callback_function_t)'
> 
>   138 |     void attachInterrupt(uint32_t channel, callback_function_t callback); // Attach interrupt callback which will be called upon compare match event of specified channel
> 
>       |          ^~~~~~~~~~~~~~~
> 
> C:\Users\admin\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.9.0\cores\arduino/HardwareTimer.h:138:10: note:   candidate expects 2 arguments, 1 provided
> 
> exit status 1
> 
> Error compiling for board 3D printer boards.
> 
> 
> 
> This report would have more information with
> "Show verbose output during compilation"
> option enabled in File -> Preferences.
> 

[image]

chrissbarr, 2020-08-24:
Hi @19504643,

In my experience, there have been issues using the Arduino IDE to compile Marlin 2.0 for STM32 boards for a while. I recommend using PlatformIO, as documented here:

https://github.com/Aus3D/RUMBA32/wiki/Marlin---Build-and-Upload-(PlatformIO)

I am not sure if your issues relate to this or not - but when I last tried, I was not able to get a successful compilation out of the IDE at all, even after all other errors were resolved. So, it's something I would recommend in general if convenient.

You are correct to have changed the board define. This is better captured on the above page, along with the info page here:

https://github.com/Aus3D/RUMBA32/wiki/Recommended-Marlin-Configuration

The instructions on configuring for the Arduino IDE are a little out of date at this stage, hence the wrong board definition.

The error that you are seeing seems to relate to the way timers are handled. Timer handling errors often relate to having the wrong version of the STM32duino Core installed (my understanding is that Marlin currently needs a slightly older version - not the latest - installed). I believe it is 1.7.0 that is needed but am not sure. PlatformIO grabs the specific version needed automatically, another reason I would recommend going that route.

I would suggest checking which version of the STM32Duino Core is installed, and changing it to v1.7.0 if possible. Otherwise, swapping over to use PlatformIO should resolve this issue automatically.

Let me know if you're still stuck, happy to look into it more.

19504643, 2020-10-12:
ok, I change back the STM32Duino Core to v1.7.0 and it works, finally I get the bin file, and flash it to the board
the LED1 is blinking, and I install the virtual comport driver, then I found the STMicroelectronics Virtual COM Port (COM10) in device manager, I open it with baudrate 250000 or 115200 in debug monitor, but it seems not working with nothing produced when it startup or I send any gcode, what's the problem