Aus3D/RUMBA32 #38: Can't connect a MKS Rumba32 board via USB

Bernouillle, 2020-10-05:
Hello there! 
I know, that the MKS Rumba32 board is not sold by your company, but maybe you can help me to solve my connection problem. I'm using Klipper as the firmware, where a Raspberry Pi is used as a host computer to do all the calculations and to communicate with the Rumba32 board. Unfortunately i am not able to find the USB connection of the Rumba32 board with the Raspberry. Do you have an idea how to fix this problem? 
Many thanks in advance!

czosnekltd, 2020-10-24:
I had the same issue with KLIPPER/MKSRumba32 and USB. Have you tried to connect via UART?

Bernouillle, 2020-10-26:
@czosnekltd i haven't tried it yet. Can you tell me how i set up the UART connection between the Pi and the MKS Rumba32? Do i have to change something in the configuration file?

czosnekltd, 2020-10-27:
```

  │ │                   [*] Enable extra low-level configuration options                                          │ │
  │ │                       Micro-controller Architecture (STMicroelectronics STM32)  --->                        │ │
  │ │                       Processor model (STM32F446)  --->                                                     │ │
  │ │                       Clock Reference (12 MHz crystal)  --->                                                │ │
  │ │                   [ ] Use USB for communication (instead of serial)                                         │ │
  │ │                   [ ] Use CAN for communication (instead of serial)                                         │ │
  │ │                       **Serial Port (USART1)  ---**>                                                            │ │
  │ │                   (250000) Baud rate for serial port                                                        │ │
  │ │                   [ ] Specify a custom step pulse duration                                                  │ │
  │ │                   ()  GPIO pins to set at micro-controller startup                                          │ │
  │ │
```
In section [MCU] in my case in RPI4:
```
[mcu]
serial: /dev/ttyS0
baud: 250000
restart_method: command
```

Bernouillle, 2020-10-28:
Thanks a lot for your help so far, @czosnekltd! I really appreciate it. I still have one question left. Which pins do i use for the UART connection on the RUMBA32? Are the TX/RX pins of the AUX connector the correct ones? And do i need a GND pin as well?

czosnekltd, 2020-10-28:
I joined:
AUX-1/AUX_TX => RPI/GPIO_15(RXT)
AUX-1/AUX_RX => RPI/GPIO_14(TXT)
AUX-1/GND => RPI/GPIO_GND 
The AUX-1 / AUXU_TX / RX outputs differ from EXP3 / USART1_TX / RX in that they are protected with additional resistors, so I think that in principle EXP-3 / USART1_TX / RX can also be connected. 
But as I wrote at the beginning, I used pins on the AUX-port for safety.