Aus3D/RUMBA32 #40: heater errors

phongshader, 2020-11-11:
I posted this  here https://github.com/makerbase-mks/MKS-RUMBA32/issues/26 but have gotten no response so I'm hoping someone here will have an answer for finding a replacement mosfet for HE0...or if that is even the problem.

chrissbarr, 2020-11-12:
Hi @phongshader,

The behaviour you're describing in the other issue there is odd, and not something that I've seen before. It's not completely clear to me if it's a hardware or a firmware problem, but if there is an issue with the MOSFET you can definitely replace it with a compatible part.

The MKS RUMBA32 boards have swapped to a different MOSFET that uses a different package to the ones in the original design. This means the replacements I would usually recommend for my RUMBA32 boards won't work, so I have to check into the MOSFET used in the MKS design a bit.

I'm not familiar with the MOSFET used, which as you mentioned is listed as "HY1403" in the MKS BOM. Based on a quick search this isn't from a manufacturer I'm familiar with, and isn't on any major distributor's websites (Digi-Key etc.), so I suspect it's from a Chinese manufacturer that mostly sells locally. 

Fortunately, there does appear to be a datasheet that can be found [here](https://datasheetspdf.com/pdf-file/1300437/HOOYI/HY1403/1).

The MOSFET looks to be a fairly common TO-252 (DPAK) package. The pinout is given in the datasheet:

[image]

So, at a bare minimum we're trying to track down something that fits that pinout.

The other important characteristics to look for would be:

- Drain-Source Voltage Rating (Vdss) >= 40V
- Gate Threshold Voltage (Vgsth) < 3V
- Drain-Source On-State Resistance (RDSon) < 30mOhm at 3V (the lower RDSon is, the less heat will be generated in the MOSFET - so lower is better, but anything under 30mOhm should do the trick). 

Interestingly the graphs in the datasheet for the HY1403 don't show the RDSon for Vgs below about 3.5V - which is an indication that it is not an ideal MOSFET for the application, as its behaviour & performance running on 3.3V logic is not defined.

(We are looking for a MOSFET with a graph like this, but where the 3V line (red) intersects with the graph and gives us an RDSon < 30mOhm):

[image]

It also has a Vdss of only 30V - which will work, but I'd be worried there's not a lot of headroom and the MOSFET could be damaged in some circumstances.

Anyway, we can use these criteria to find a better replacement! After a quick search on Digi-Key, I think the [BUK9217-75B](https://www.digikey.com/en/products/detail/nexperia-usa-inc/BUK9217-75B118/1163571) looks like a good candidate (values from datasheet):

Vdss = 75V
Vgsth < 2.3V
RDSon @ 3.3V ~= 24mOhm (reading below plot)

We can compare the RDSon vs Vgs plot against the one for the HY1403:

[image]

We can see that the plot shows the behaviour at 3V is defined, so we can feel comfortable with how the MOSFET will behave in our application.

Finally, we can check the pinout:

[image]

Though the name is different, SOT428 and TO-252 are both the same package (DPAK). The pinout here is the same - gate, drain, source, from left to right.

So, I would suggest the [BUK9217-75B](https://www.digikey.com/en/products/detail/nexperia-usa-inc/BUK9217-75B118/1163571) is a suitable replacement. There are probably plenty of other parts that would do just as well - but I can't see any reason this would be a bad choice, and it should be much better than the default part used. If you can't source one of these, let me know and I'll see if I can find any other suggestions.

Hope that helps!

phongshader, 2020-11-21:
WOW!! Thank you for the info