Aus3D/RUMBA32 #41: Where to connect AD595 thermocouple pcb???

Hobby4life, 2020-11-11:
I am trying to connect a AD595 thermocouple pcb.

TEMP_SENSOR_0 -1

On the old 2560 board it could be connected to EXP3.
and in the rumba pins.h file there was a defenition for using TEMP_SENSOR_0 -1 (pin 6)

Now on the Rumba32 I cant seem to find any info how and where to connect the thermocouple pcb.

Any help would be apreciated

chrissbarr, 2020-11-12:
Hi @Hobby4life,

This will be a case of yes, but...

You should be able to use the same pin on the EXP3 connector. What you're looking for is a pin that can be used as an ADC input (analog input). RUMBA+ and RUMBA32 both have two ADC pins on the EXP3 connector, so that they're compatible with most wiring in common.

The pinout for the connector is shown here:

[image]

The ADC pins are PA3 and PA4. Either pin should work, you will just need to define the matching pin in the firmware settings.

In firmware, you'll want to:
- set `TEMP_SENSOR_0` to -1, as you're doing
- change the `TEMP_0_PIN` pin in the [pins_RUMBA32_common.h file](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/pins/stm32f4/pins_RUMBA32_common.h#L109) to PA3 (or PA4) to match your wiring.

_**However**, I would recommend against using the AD595_ with the RUMBA32 board. AD595 is designed to run on 5V logic, and is capable of driving the ADC output up to 5V and damaging the RUMBA32 board. Because the output voltage is proportional to the thermocouple temperature, this would happen at higher temperatures - so you might damage the RUMBA32 board if the temperature goes above say 300 degrees or so. 

It looks like the AD8495 is a similar part that runs on 3.3V and is supported in Marlin. If possible, I would recommend changing to a board using that amplifier instead.

If you do want to use the AD595, I would recommend adding a protection circuit between the AD595 and the RUMBA32 board - something like the diode clipping circuit shown [here](https://electronics.stackexchange.com/a/163467).

Hobby4life, 2020-11-12:
Hi,Funny I indeed discovered the same today, PA4 is the pin I need. PA3 behaves like a PWM pin,, so got no luck there.

Indeed about the AD595 runs on 5V and its reference too.. 

The AD8495 will be the best choice.

Thanks!

Hobby4life, 2020-11-16:
I have received a Adafruit AD8495 board last week.

Got it to work but.........

The Adafruit board uses an internal reference voltage of 1.25v

In the temperature.cpp file the calculation for the AD8495 is not correct.
A. Where the heck did 6.6 come from? I gues a typo error instead of 3.3??
B. In the default calculation there is no reference voltage taken in.

#define TEMP_AD595(RAW)  ((RAW) * 5.0 * 100.0 / float(HAL_ADC_RANGE) / (OVERSAMPLENR) * (TEMP_SENSOR_AD595_GAIN) + TEMP_SENSOR_AD595_OFFSET)
//#define TEMP_AD8495(RAW) ((RAW) * 6.6 * 100.0 / float(HAL_ADC_RANGE) / (OVERSAMPLENR) * (TEMP_SENSOR_AD8495_GAIN) + TEMP_SENSOR_AD8495_OFFSET)

uncommented the AD8495 line.

And add this line.
It uses the correct ADC voltage range of 3.3 for calculation and calculates the 1.25v reference voltage of the AD8495 Adafruit board.

#define TEMP_AD8495(RAW) ((((RAW) * 3.3 / 1024.0 / (OVERSAMPLENR) - 1.25) / 0.005 / (TEMP_SENSOR_AD8495_GAIN)) + TEMP_SENSOR_AD8495_OFFSET)