Aus3D/RUMBA32 #43: Board availability

CryoRig, 2020-11-19:
Hi,

is there any information on when the Board will be available again through Aus3D?
The Shop has been showing nothing in stock for at least a month.
I tried to contact Aus3D via their Contact form but haven`t heard anything back...

CryoRig, 2020-11-20:
Closing because i found a german retailer with v1.1b stock (youprintin3d.de)(i am not affiliated with them)

closed 2020-11-20 by CryoRig

McPrince96, 2020-12-30:
@CryoRig did you receive it yet?

I also found this one a couple days ago and i was wondering if they deliver a good product and service?

CryoRig, 2020-12-30:
@McPrince96 yes i got one, works fine
service was fine, had a question about the board version and got an answer 3 or 4 hours later