Aus3D/RUMBA32 #44: Ist it possible to connect an In-Circuit-Debugger?

JoeScotty, 2020-11-26:
Hi, just a question:
Just testing and doing some small mods using VSC and platformio. 
Sometimes it would be more efficient compared to lot of log outputs.

Ist it possible to connect an In-Circuit-Debugger to the Rumba32
like ST-Link or Black Magic Probe, which are using either JTAG or Serial Wire Mod Interfrace ?
Thanks!

chrissbarr, 2020-11-27:
Hi @JoeScotty,

That should definitely be possible, it will come down to what pins you need access to given your debugger and settings etc.

I have broken out the most common debug pins to a 6-pin header, labelled "SWD". You can see it here:

[image]

That header gives you access to:

- SWDIO
- SWCLK
- 3.3V
- GND
- NRST
- BOOT0

I believe that SWDIO and SWCLK are sufficient for debugging with STM32-compatible SWD tools. I did some debugging using an ST-Link at one point during development (2+ years ago now), and don't remember having to attach any additional wires etc., so I think that it is doable with the above.

JoeScotty, 2020-11-27:
Hi Chris,
thank you very much for your quick reply!

JoeScotty, 2020-12-19:
Hi Chris,
after some trial and errors I managed to start a debug session and just wanted to share.
Your hints were correct, I'll specify a bit deeper:

**Coding environment (I'm no software developer, please excuse any miswording)**
- OS is Win10
- Rumba32 Board V1.03e
- VSCode with PlatformIO
- repetier-Firmware dev2

**Hardware**
- STLink V2 Programmer: ST Link V2 Mini Simulator Download Programmer (STM8 & STM32)
https://www.az-delivery.de/en/products/st-link-v2-mini-simulator-download-programmer-stm8-6-stm32?_pos=1&_sid=28054885e&_ss=r
- Other programmers with ST-Link V2 and STM32 should be functional as well.

**Cable:**
- with the programmer comes a 4 Pin ribbon cable with 4 single sockets at both ends.
- wiring diagram:
```
STLink                     Rumba32
Pin name    pin #          Pin name    pin #    
SWCLK       2        -      SWCLK       3
SWDIO       4        -      SWDIO       1
GND         6        -      GND         6
3V3         8        -      3V3         2
```

> In my case I assembled a 10-pin ribbon cable using a 10-pin-socket and a 6-pin-socket.

- Pinout of STLink V2: View to Pin Tips
[image]

- Pinout of Rumba32, View to Component side of board

[image]

- Schematics detail of Rumba32
[image]

**PlatformIO Settings in platformio.ini:**

- the previous rumba 32 section contained:
```
[env:RUMBA32]
....
upload_protocol = dfu
```
- change to :
```
[env:RUMBA32]
....
upload_protocol = stlink
debug_tool = stlink
```

**Connection to Desktop**
- For the STLink V2 a driver is necessary, goto
https://www.st.com/en/development-tools/stsw-link009.html#get-software
enter name and email, then the driver can be downloaded.
Install with admin rights.

- either 2 USB Cables, one to micro USB of Rumba32 board, the other one to the ST Link Programmer.

- also tried one USB cable and an USB Hub clode to the printer. (Once i had a strange behaviour, but could not reproduce)
- It was better to firstly switch on the power supply of the rumba board and afterwards plug in USB.
- after plugging in both USB connectors, the device manager shows
 COMx under COM&LPT (like before) and - new - "STM32 STLink" under USB-Devices.
 
And - by the way - using the ST Link method to update firmware does not require pressing buttons on the board,
which is helpful in case the board is mounted deeply in the printer.

Hope this helps somebody else!