Aus3D/RUMBA32 #52: RUMBA32 Aus3D V1.0 Not appearing under devices in boot mode

alanamer, 2023-10-20:
Hello,

I am trying to upload Marlin to an AUS3D RUMBA32 V1.00 using PlatformIO in Visual Studio Code on a Mac. When the board is not in boot mode, I can see it under devices "RUMBA32_F446VE CDC in FS Mode," however, once it enters boot mode, it is no longer findable. I have tried adding the 100K resistor recommended here[https://github.com/Aus3D/RUMBA32/issues/34]() but that did not resolve the issue. I am entering boot mode by holding boot and then pressing reset, then releasing. There is one flash and then the LED turns off. 

Right now, if I try to upload when the board is not showing up, I receive this error: *** [.pio/build/rumba32_f446ve/firmware.bin] Error -6.

This is my platformio.ini configuration:
[env:rumba32_f446ve]
platform = ststm32
board = rumba32_f446ve
framework = arduino
upload_protocol = dfu

I would really appreciate any ideas on how to solve this! Thank you!

alanamer, 2023-10-23:
So I got the firmware to upload through platform ino, but now there is no LED on the board and I am unable to find it as a device in either Pronterface or studio code

coolcheat, 2024-01-01:
I'm using the stlink method, which is a lot more reliant as well as being faster.

You can see here:
https://github.com/Aus3D/RUMBA32/issues/44